syssi/esphome-tianpower-bms #1: Add Tianpower TP-LT55 support (used in BASEN 48V 200Ah battery packs)

xpinguinx, 2023-06-03:
Is it possible to develop this same ESPHome component to monitor a 48Volt Basen BMS via the Bluetooth port and receive data in MQTT??

I tried to use "esp32-ble-example.yaml" and through the MQTT-EXPLORER debug application I see that the ESP32 connects to the BMS:

""**[0;35m[C][mqtt.switch:041]: MQTT Switch 'basen-bms-ble enable bluetooth connection': [0m**

but after a few seconds it disconnects displaying this other error: 
**[0;33m[W][basen_bms_ble:213]: [A4:C1:38:27:48:9A] Not connected [0m**

I note that the "switch" topics are displayed correctly in the MQTT brocker while the "sensors" of the cells, temperatures and everything else are not displayed.

I trust in your help, thank you

xpinguinx, 2023-06-03:
I absolutely have to solve the bluetooth communication problem by exporting the data and sensors in the MQTT format.
Can you please tell me what I should do and how long it takes for this job? 
Can we talk in private? 
Thank you

syssi, 2023-06-03:
I've pushed a first draft of the tianpower protocol implementation. Please flash this YAML

https://github.com/syssi/esphome-tianpower-bms/blob/main/esp32-ble-example.yaml

and provide a ESPHome log. The log can be retrieve using the ESPhome CLI:

```
esphome logs esp32-ble-example.yaml
```

syssi, 2023-06-03:
Could you check the Android app. Are there switches / button to control the mosfet state (charging on/off, discharging on/off)? Could you provide some screenshots of the different views of the Android app?

xpinguinx, 2023-06-03:
> Puoi controllare l'app per Android. Ci sono interruttori/pulsanti per controllare lo stato del mosfet (carica on/off, scarica on/off)? Puoi fornire alcuni screenshot delle diverse visualizzazioni dell'app per Android?

[image]
[image]
[image]
[image]
[image]

[logs_tianpowerbms_compile.txt](https://github.com/syssi/esphome-tianpower-bms/files/11643028/logs_tianpowerbms_compile.txt)

xpinguinx, 2023-06-03:
> Could you check the Android app. Are there switches / button to control the mosfet state (charging on/off, discharging on/off)? Could you provide some screenshots of the different views of the Android app?

[Image 1.zip](https://github.com/syssi/esphome-tianpower-bms/files/11643040/Image.1.zip)
[Image 2.zip](https://github.com/syssi/esphome-tianpower-bms/files/11643041/Image.2.zip)
[Image 3.zip](https://github.com/syssi/esphome-tianpower-bms/files/11643042/Image.3.zip)
[Image 4.zip](https://github.com/syssi/esphome-tianpower-bms/files/11643043/Image.4.zip)
[logs_tianpowerbms_compile.zip](https://github.com/syssi/esphome-tianpower-bms/files/11643044/logs_tianpowerbms_compile.zip)

syssi, 2023-06-03:
I've pushed a fixed version of the YAML configuration.

syssi, 2023-06-03:
The current implementation should be able to connect to the BMS and subscribe to some notification streams. The received raw data should be printed to the log. Please provide such a log. As soon as we are sure everything works as expected we can move on.

xpinguinx, 2023-06-03:
fails to finish compiling "esp32-ble-example.yaml" (error log file attached)

[logs_tianpowerbms_compile (1).txt](https://github.com/syssi/esphome-tianpower-bms/files/11643263/logs_tianpowerbms_compile.1.txt)

syssi, 2023-06-03:
A clean build is required:

```
esphome clean esp32-ble-example.yaml
esphome run esp32-ble-example.yaml
```

If you use the ESPHome dashboard there should be a "clean build" button.

xpinguinx, 2023-06-03:
> 

...with which procedure and how should I export the logs?
debug MQTT I see this result

[image]

syssi, 2023-06-03:
Do you use the ESPHome dashboard? If you click on a ESPHome node and choose the "log" button you should be able to copy&paste the log. A log capture of around 1-3 minutes would be nice.

xpinguinx, 2023-06-03:
> Usi la dashboard di ESPHome? Se fai clic su un nodo ESPHome e scegli il pulsante "log", dovresti essere in grado di copiare e incollare il log. Un'acquisizione del registro di circa 1-3 minuti sarebbe utile.

Attached the log. Good work :)

[logs.txt](https://github.com/syssi/esphome-tianpower-bms/files/11643423/logs.txt)

syssi, 2023-06-03:
I've pushed another change. In best case the component is able to decode and publish some measurements now via MQTT. Could you flash your device again and provide another log?

xpinguinx, 2023-06-03:
Unfortunately nothing has changed in MQTT :(

[image]

I attach the LOG file in which it seems to me that there is an error:
[logs_bms_compile.txt](https://github.com/syssi/esphome-tianpower-bms/files/11644158/logs_bms_compile.txt)

syssi, 2023-06-03:
I've fixed the mentioned issue. Please compile & flash the most recent version and provide another log. The log should provide the reason for not pushing measurements. These topics are implemented:

```
tianpower-bms-ble/sensor/tianpower-bms-ble_state_of_charge/state
tianpower-bms-ble/sensor/tianpower-bms-ble_total_voltage/state
```

xpinguinx, 2023-06-03:
log updated

[logs_bms_compile.txt](https://github.com/syssi/esphome-tianpower-bms/files/11644265/logs_bms_compile.txt)

syssi, 2023-06-04:
I've pushed a fix. Please give it another try!

xpinguinx, 2023-06-04:
Good morning Syssi,
Updated register, I remain available
[logs_bms_compile_4.txt](https://github.com/syssi/esphome-tianpower-bms/files/11645000/logs_bms_compile_4.txt)

I also attach the MQTT status
[image]

syssi, 2023-06-04:
The BMS wasn't available this time. May be because it supports just one simultaneous connection and the BLE module was too busy to pick up the new connection:

```
[W][tianpower_bms_ble:126]: [90:A6:BF:93:A0:69] Not connected
[W][tianpower_bms_ble:126]: [90:A6:BF:93:A0:69] Not connected
[W][tianpower_bms_ble:126]: [90:A6:BF:93:A0:69] Not connected
[W][tianpower_bms_ble:126]: [90:A6:BF:93:A0:69] Not connected
```

Could you provide another log?

xpinguinx, 2023-06-04:
WOW :)

[image]

Updated register:
[logs_bms_compile_5.txt](https://github.com/syssi/esphome-tianpower-bms/files/11645176/logs_bms_compile_5.txt)

syssi, 2023-06-04:
LOL. All measurements are a bit off. ;-) Will be fixed soon.

xpinguinx, 2023-06-04:
> LOL. Tutte le misurazioni sono un po' fuori. ;-) Verrà risolto presto.

When you have finished the configuration, let me know your paypal address, thanks.

syssi, 2023-06-04:
The decoding of the measurements is fine now. I would be happy if you could apply some load to your battery so you could check the `current` sensor is working properly. If there is load this value should be positive. If you charge the battery the value should be negative. Please double check the resolution of the sensor. The factor is probably wrong at the moment.

xpinguinx, 2023-06-04:
Dear Syssi,
finally the goal was a success thanks to your availability!

At the moment I don't have the possibility to check the "current" sensor because the photovoltaic system is in production with the old BMS.
We remain that if I encounter problems I will update you in a few weeks.

Working with you has been pleasant and for this I congratulate you also for your professionalism and knowledge.

syssi, 2023-06-04:
You are welcome! I tried to contact you via bms@.

closed 2023-06-05 by syssi