syssi/esphome-tianpower-bms #10: Reading for current and power incorrect

thezjo, 2023-12-27:
See below, I get strange readings for power; charging power; discharging power and current, rest of the readings are OK.
Im loading with 30 Amps and discharging power is about 1500 Watt.
Thanks for your help

[15:37:17][I][tianpower_bms_ble:307]: Status frame received
[15:37:17][D][tianpower_bms_ble:308]:   55.14.83.00.27.14.3E.00.BE.00.96.00.D2.F4.6B.30.30.00.64.AA (20)
[15:37:17][D][sensor:094]: 'bms state of charge': Sending state 39.00000 % with 0 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms total voltage': Sending state 51.82000 V with 2 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms average temperature': Sending state 19.00000 °C with 1 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms ambient temperature': Sending state 15.00000 °C with 1 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms mosfet temperature': Sending state 21.00000 °C with 1 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms current': Sending state -2965.00000 A with 3 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms power': Sending state -153646.29688 W with 2 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms charging power': Sending state 0.00000 W with 2 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms discharging power': Sending state 153646.29688 W with 2 decimals of accuracy
[15:37:17][D][sensor:094]: 'bms state of health': Sending state 100.00000 % with 0 decimals of accuracy
[15:37:17][D][tianpower_bms_ble:600]: Send command (handle 0x15): 55.04.84.AA

syssi, 2023-12-27:
Thanks for your bug report and log capture. I will try to reproduce the issue using the raw traffic of the status frame above. In best case it can be fixed easily.

thezjo, 2023-12-27:
Thanks Syssi,

I found one more problem with the max and min cel voltages.
In the case below, the bms min cell voltage = 3.256 and bms max cell voltage  = 3.260 based on the 16 individual cell readings.

[19:37:38][I][tianpower_bms_ble:458]: Cell voltages frame (chunk 0) received
[19:37:38][D][tianpower_bms_ble:459]:   55.14.88.0C.BC.0C.BB.0C.BC.0C.BC.0C.B9.0C.B8.0C.BB.0C.BC.AA (20)
[19:37:38][D][sensor:094]: 'bms cell voltage 1': Sending state 3.26000 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 2': Sending state 3.25900 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 3': Sending state 3.26000 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 4': Sending state 3.26000 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 5': Sending state 3.25700 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 6': Sending state 3.25600 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 7': Sending state 3.25900 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 8': Sending state 3.26000 V with 3 decimals of accuracy
[19:37:38][D][tianpower_bms_ble:600]: Send command (handle 0x15): 55.04.89.AA
[19:37:38][I][tianpower_bms_ble:458]: Cell voltages frame (chunk 1) received
[19:37:38][D][tianpower_bms_ble:459]:   55.14.89.0C.BB.0C.BC.0C.BB.0C.BA.0C.BA.0C.BB.0C.BB.0C.BC.AA (20)
[19:37:38][D][sensor:094]: 'bms cell voltage 9': Sending state 3.25900 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 10': Sending state 3.26000 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 11': Sending state 3.25900 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 12': Sending state 3.25800 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 13': Sending state 3.25800 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 14': Sending state 3.25900 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 15': Sending state 3.25900 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms cell voltage 16': Sending state 3.26000 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms min cell voltage': Sending state 3.15400 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms max cell voltage': Sending state 3.29600 V with 3 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms max voltage cell': Sending state 16.00000  with 0 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms min voltage cell': Sending state 9.00000  with 0 decimals of accuracy
[19:37:38][D][sensor:094]: 'bms delta cell voltage': Sending state 0.14200 V with 3 decimals of accuracy

syssi, 2023-12-27:
Could you create another issue for cell voltages bug? Thanks in advance!

syssi, 2023-12-27:
I've fixed the decimal place of the current sensor. This should solve this issue. Please compile & flash you ESP another time and give it a try!

commit 617a2faa (2023-12-27): Fix decimal place of the current sensor (Closes: #10) (#11)
closed 2023-12-27 by syssi