syssi/esphome-tianpower-bms #17: BOX BASEN 280 ah BMS Tianpower

PASAR78, 2024-04-25:
Hi,
I installed the project on HA, changed the mac address, the BT of the box is TP_123456, the log cannot connect.
The BOX BASEN 280/300 ah is as follows
https://www.basengreen.com/it/PRODOTTI/custodia-fai-da-te-per-accessori-per-contenitori-per-batterie-48v-300ah/

I attach the log.
[logs_esp32-bms-basen_logs.txt](https://github.com/syssi/esphome-tianpower-bms/files/15103692/logs_esp32-bms-basen_logs.txt)

Thanks for your answer

syssi, 2024-04-25:
Could you increase the log level and provide another log? Thanks in advance!

https://github.com/syssi/esphome-tianpower-bms?tab=readme-ov-file#debugging

PASAR78, 2024-04-28:
hi,
the debugger log has been increased, the BT does not connect to the BMS, I attach the updated log.
last night it was connected for a few minutes, then it disconnected, see image.
Thanks in advance!
[logs_esp32-bms-basen_logs (1).txt](https://github.com/syssi/esphome-tianpower-bms/files/15141303/logs_esp32-bms-basen_logs.1.txt)
[image]

would there be the possibility of connecting the BMS with the RS485 port and therefore with the modbus? I have already connected the inverter this way.

syssi, 2024-04-28:
Note for myself:

> Tianpower TP-LT55AT used in BASEN 48V 280Ah Bluetooth battery bank
> Advertised via BLE as TP_123456

syssi, 2024-04-28:
> would there be the possibility of connecting the BMS with the RS485 port and therefore with the modbus? I have already > connected the inverter this way.

I must admit: I don't know much about this BMS. Could you ask your seller about some RS485 protocol details? Do you have a RS485 converter and could you sniff some data?

PASAR78, 2024-04-29:
> What's the distance between the BMS and the ESP32? Which ESP32 model/version do you use? Could you try to replace the ESP32 or change some more parameter (f.e. more distance vs. less distance to the BMS).

I purchased an ESP32-WROOM-32D board purchased on AliExpress, I can try to install the integration on another ESP32, for now I have tried moving it further away (always in the same room) but it cannot connect.

PASAR78, 2024-04-29:
> I must admit: I don't know much about this BMS. Could you ask your seller about some RS485 protocol details? Do you have a RS485 converter and could you sniff some data?

the seller of the box basically mounts the Tianpower BMS but doesn't know anything else.
I have a RS485 converter and I could read the parameters, if you tell me which program to use, I can try to sniff the data.

PASAR78, 2024-05-07:
> I purchased an ESP32-WROOM-32D board purchased on AliExpress, I can try to install the integration on another ESP32, for now I have tried moving it further away (always in the same room) but it [...]

I did the test with another ESP32 but it still doesn't connect to the BT of the BMS, at this point I think the problem is with the BT card and for this reason I would like to try to read the data with the RS485 port.

closed 2024-05-10 by syssi

PASAR78, 2024-05-13:
HI
I wanted to let you know that I was able to make the integration work.
I used an ESP32 board with CP2102 and restarted the box in order to associate the BT of the BMS.
There are some parameters that give me zero and small things to fix, if you want I'm available to send you the logs, you could make an integration calling it specific for the BASEN bms installed on the BOXES.

syssi, 2024-05-13:
Cool! Thanks for your feedback. Feel free to create a new issue and provide some logs + contribute improvements if you can.