syssi/esphome-tianpower-bms #21: 2 Basen battery

italodl, 2024-07-14:
hi, I have 2 basen batteries and two esp32. I can't configure the second battery. the sensors have the same name and do not work. Is it possible to make both work?

syssi, 2024-07-15:
Could you increase the log level

```
logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble: DEBUG
    esp32_ble_tracker: VERY_VERBOSE
    tianpower_bms_ble: VERY_VERBOSE
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

and provide some logs of both batteries?

syssi, 2024-07-16:
Could you provide some more details? For example are the ESP32 models different? It looks like ESP32S and ESP32-C3 models doesn't to a good job on Bluetooth at the moment.

syssi, 2024-07-16:
If the naming of the sensors the only issue and the data arrives properly already you have to make sure the ESPHome nodes has different names:

```
# first node

substitutions:
  name: tianpower-bms-ble

# ...

sensor:
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
```

```
# second node

substitutions:
  name: tianpower-bms-ble2

# ...

sensor:
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
```

The `name` is part of the entities.

italodl, 2024-07-17:
I used the same version of eps32 but I used the same name. I will use different names in the next tests and I will let you know the outcome

syssi, 2024-07-22:
Let's close the issue. Feel free to re-open the issue if it still doesn't work.

closed 2024-07-22 by syssi

italodl, 2024-07-22:
hi, I tried changing the name and now the two batteries are monitored perfectly, each with the respective esp32
thanks very much

syssi, 2024-07-22:
Awesome! Have fun!