syssi/esphome-tianpower-bms #25: Add new protocol (SOF `0x7B`, EOF `0x7D`) support

syssi, 2024-07-23:
Start of frame `0x7B`
End of frame `0x7D`

```
# Requests

0x7B64007D: Acknowledgment after receiving device list
0x7B01007D: Request for basic device information
0x7B02007D: Request for additional device information
0x7B03007D: Request for more device information
0x7B08007D: Request for historical data
0x7BF3007D: Request for specific device information
0x7BF4007D: Request for more specific device information
0x7BF5007D: Request for even more specific device information
0x7B09007D: Request for manufacturing date
0x7B04007D: Request for parameter data set 1
0x7B05007D: Request for parameter data set 2
0x7B06007D: Request for parameter data set 3
0x7B07007D: Request for parameter data set 4
0x7B0A007D: Request for additional parameter data set 1
0x7B0B007D: Request for additional parameter data set 2
0x7B0C007D: Request for additional parameter data set 3
0x7BE2007D: File transfer completion request
0x7B81007D: Request for number of cycles information

# Responses

0x7B73: Device discovery response
0x7B74: Online device list response
0x7BFA: Master/Slave setting response
0x7BFB: Clear historical data response
0x7BFC: General setting response (success/failure)
0x7B01: Basic device information response
0x7B02: Additional device information response
0x7B03: More device information response
0x7B08: Historical data response
0x7BF3, 0x7BF4, 0x7BF5: Various device information responses
0x7B09: Manufacturing date response
0x7B04, 0x7B05, 0x7B06, 0x7B07: Parameter data responses
0x7B0A, 0x7B0B, 0x7B0C: More parameter data responses
0x7BE0, 0x7BE1, 0x7BE2: File transfer handshake and data transmission responses
0x7BF9: Parameter setting response
0x7BFD: Another parameter setting response
0x7B70: Restore BMS default parameters response
0x7B71: Historical data reading response
0x7B72: Delete BMS historical data response
0x7B91: Set parameters response
0x7B81: Number of cycles information response
0x7B2F: Unknown response (triggers parameter data request if successful)
0x7B15: Unknown response (triggers parameter data request if successful)
0x7B62: DDA2 string response
```

kopierschnitte, 2025-05-04:
Great! Then I'll try to repeat the capture while writing down the corresponding values. I guess that would help ;-)

For now I can already tell you that it's a 300Ah pack (16S) with a SoC between 60 and 70%.

syssi, 2025-05-04:
Coud you point me to the APK file of the Android App (version) supporting this BMS?

kopierschnitte, 2025-05-04:
Of course. There are several almost identical ones but I'm using https://www.basenpower.com/wp-content/uploads/2024/11/Bluetooth.apk

But I have to admit that it's obfuscated as hell :-(

kopierschnitte, 2025-05-04:
I've also found an obvious firmware updater which looks almost identical. Maybe you'll find something interesting there: https://play.google.com/store/apps/details?id=com.jizhi.beiliUpgradefixed

syssi, 2025-05-04:
It's not verified/perfect but a first attempt to decode the frames:

# Frame Descriptions

## 0x7b01 response

| Byte Position | Length (Bytes) | Field | Description |
|--------------|--------------|------|-------------|
| 3-4 | 2 | RSOC | Relative State of Charge |
| 5-6 | 2 | totalVoltage | Total battery voltage |
| 11-12 | 2 | moswd | MOSFET temperature |
| 13-14 | 2 | electricCurrent | Electric current |
| 15-16 | 2 | remainingCapacity | Remaining capacity |
| 17-18 | 2 | fullCapacity | Full capacity |
| 19-20 | 2 | nominalCapacity | Nominal capacity |
| 21-22 | 2 | xhrl | XHRL value |
| 23-24 | 2 | numberOfCycles | Number of cycles |
| 25-26 | 2 | equilibriumState | Equilibrium state |
| 27-28 | 2 | equilibriumStateHight | Upper equilibrium state |
| 29-30 | 2 | FET | FET control status |
| 31-32 | 2 | protectionStatus | Protection status |
| 33-34 | 2 | SOH | State of Health (if bytes 2-2.5 is not 30) |

## 0x7b02 response

| Byte Position | Length (Bytes) | Field | Description |
|--------------|--------------|------|-------------|
| 2-2.5 | 1 | - | Used to calculate the number of voltage values |
| 3-3.5 | 1 | batteryNum | Number of batteries |
| 4+ | 2 per cell | monomerVoltage[i] | Voltage values of individual cells |

## 0x7b03 response

| Byte Position | Length (Bytes) | Field | Description |
|--------------|--------------|------|-------------|
| 3-3.5 | 1 | NtcNum | Number of temperature sensors |
| 4+ | variable | NtcInfo | NTC temperature information |

## 0x7b08 response

| Byte Position | Length (Bytes) | Field | Description |
|--------------|--------------|------|-------------|
| 3-4 | 2 | scpt | SCPT value |
| 5-6 | 2 | cot | COT value |
| 7-8 | 2 | dot | DOT value |
| 9-10 | 2 | otom | OTOM value |
| 11-12 | 2 | tomu | TOMU value |
| 13-14 | 2 | oot | OOT value |
| 15-16 | 2 | out | OUT value |
| 17-18 | 2 | chtt | CHTT value |
| 19-20 | 2 | ltct | LTCT value |
| 21-22 | 2 | htdt | HTDT value |
| 23-24 | 2 | ltdt | LTDT value |

## 0x7bf3 response

| Byte Position | Length (Bytes) | Field | Description |
|--------------|--------------|------|-------------|
| 3-3.5 | 1 | softwareVersion (Main) | Main version number of software |
| 4-4.5 | 1 | softwareVersion (Sub) | Sub version of software |

## 0x7bf4 response

| Byte Position | Length (Bytes) | Field | Description |
|--------------|--------------|------|-------------|
| 3-3.5 | 1 | BmsVersion (Main) | Main version number of BMS |
| 4-4.5 | 1 | BmsVersion (Sub) | Sub version of BMS |
| 5-5.5 | 1 | BmsVersion (Build) | Build number of BMS |

## 0x7bf5 response
[... 109 lines cut ...]
## 0x7b64 response

| Byte Position | Length (Bytes) | Field | Description |
|--------------|--------------|------|-------------|
| 3-3.5 | 1 | Status | Status bits (interpreted as binary string) |

## 0x7b71 response

| Byte Position | Length (Bytes) | Field | Description |
|--------------|--------------|------|-------------|
| 4-4.5 | 1 | Year | Year (+ 2000) |
| 5-5.5 | 1 | Month | Month |
| 6-6.5 | 1 | Day | Day |
| 7-7.5 | 1 | Hour | Hour |
| 8-8.5 | 1 | Minute | Minute |
| 9-9.5 | 1 | Second | Second |
| 10-11 | 2 | totalBatteryVoltage | Total battery voltage (divided by 10) |
| 12-13 | 2 | batteryPackCurrent | Battery current (divided by 10) |
| 26-27 | 2 | SOH | State of Health (divided by 10) |
| 28-29 | 2 | SOC | State of Charge (divided by 10) |
| 31-32 | 2 | ratedCapacity | Rated capacity |
| 33-34 | 2 | ambientTemperature | Ambient temperature (divided by 10) |
| 35-36 | 2 | MOSTubeTemperature | MOSFET temperature (divided by 10) |
| 45-45.5 | 1 | Count | Number of voltage measurements |
| 46+ | variable | - | Voltage values for max/min determination |

syssi, 2025-05-04:
Frame `7B:01:20:00:45:14:8D:00:DC:00:B4:00:B4:00:00:52:05:75:30:75:30:00:00:00:01:00:00:00:00:00:0C:00:00:00:64:7D` decoded:

| Byte Position | Hex Value | Field | Decoded Value |
|--------------|-----------|------|---------------|
| 0-1 | 7B:01 | Frame header | 0x7B01 (frame identifier) |
| 2-2.5 | 20 | Frame length | 0x20 (32 bytes) |
| 3-4 | 00:45 | RSOC | 69% (0x0045) |
| 5-6 | 14:8D | totalVoltage | 5261 mV (0x148D) |
| 7-10 | 00:DC:00:B4 | (not in table) | - |
| 11-12 | 00:B4 | moswd | 180 (0x00B4) |
| 13-14 | 00:00 | electricCurrent | 0 A (0x0000) |
| 15-16 | 52:05 | remainingCapacity | 21253 mAh (0x5205) |
| 17-18 | 75:30 | fullCapacity | 30000 mAh (0x7530) |
| 19-20 | 75:30 | nominalCapacity | 30000 mAh (0x7530) |
| 21-22 | 00:00 | xhrl | 0 (0x0000) |
| 23-24 | 00:01 | numberOfCycles | 1 cycle (0x0001) |
| 25-26 | 00:00 | equilibriumState | 0 (0x0000) |
| 27-28 | 00:00 | equilibriumStateHight | 0 (0x0000) |
| 29-30 | 00:0C | FET | 12 (0x000C) |
| 31-32 | 00:00 | protectionStatus | 0 (0x0000) |
| 33-34 | 00:64 | SOH | 100% (0x0064) |
| 35 | 7D | Frame end | 0x7D (frame terminator) |

Frame `7B:02:21:10:0C:D8:0C:DA:0C:D9:0C:D8:0C:D8:0C:D8:0C:D9:0C:D8:0C:D8:0C:D9:0C:D9:0C:D8:0C:D9:0C:D9:0C:D8:0C:D6:7D` decoded:

| Byte Position | Hex Value | Field | Decoded Value |
|--------------|-----------|------|---------------|
| 0-1 | 7B:02 | Frame header | 0x7B02 (frame identifier) |
| 2-2.5 | 21 | Frame length | 0x21 (33 bytes) |
| 3-3.5 | 10 | batteryNum | 16 batteries (0x10) |
| 4-5 | 0C:D8 | monomerVoltage[1] | 3288 mV (0x0CD8) |
| 6-7 | 0C:DA | monomerVoltage[2] | 3290 mV (0x0CDA) |
| 8-9 | 0C:D9 | monomerVoltage[3] | 3289 mV (0x0CD9) |
| 10-11 | 0C:D8 | monomerVoltage[4] | 3288 mV (0x0CD8) |
| 12-13 | 0C:D8 | monomerVoltage[5] | 3288 mV (0x0CD8) |
| 14-15 | 0C:D8 | monomerVoltage[6] | 3288 mV (0x0CD8) |
| 16-17 | 0C:D9 | monomerVoltage[7] | 3289 mV (0x0CD9) |
| 18-19 | 0C:D8 | monomerVoltage[8] | 3288 mV (0x0CD8) |
| 20-21 | 0C:D8 | monomerVoltage[9] | 3288 mV (0x0CD8) |
| 22-23 | 0C:D9 | monomerVoltage[10] | 3289 mV (0x0CD9) |
| 24-25 | 0C:D9 | monomerVoltage[11] | 3289 mV (0x0CD9) |
| 26-27 | 0C:D8 | monomerVoltage[12] | 3288 mV (0x0CD8) |
| 28-29 | 0C:D9 | monomerVoltage[13] | 3289 mV (0x0CD9) |
| 30-31 | 0C:D9 | monomerVoltage[14] | 3289 mV (0x0CD9) |
| 32-33 | 0C:D8 | monomerVoltage[15] | 3288 mV (0x0CD8) |
| 34-35 | 0C:D6 | monomerVoltage[16] | 3286 mV (0x0CD6) |
| 36 | 7D | Frame end | 0x7D (frame terminator) |

kopierschnitte, 2025-05-04:
Wow, may I ask if you really found something in the decompiled sources or is it just your experience with these kind of devices?

kopierschnitte, 2025-05-04:
> Frame `7B:01:20:00:45:14:8D:00:DC:00:B4:00:B4:00:00:52:05:75:30:75:30:00:00:00:01:00:00:00:00:00:0C:00:00:00:64:7D` decoded:

The decoded values are correct. At least those I could verify immediately (SoC, voltages, load, capacity)

Very impressive!

syssi, 2025-05-04:
> Wow, may I ask if you really found something in the decompiled sources or is it just your experience with these kind of devices?

Did you see the javascript bundle at the APK?

syssi, 2025-05-05:
I just found this article: https://dou-ua.translate.goog/forums/topic/42715/?_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=wapp

kopierschnitte, 2025-05-05:
Wow, great article. Thanks for sharing!

Is there anything I can do to help you with development?

syssi, 2025-05-05:
The next part is to identify a good name for the protocol implemention. IMO this isn't a Tianpower device.

kopierschnitte, 2025-05-05:
Yes, this is tricky. The BT-Name indicates a CERRNSS device but the firmware starts with KS.

I would suggest going the CERRNSS route because it could be easily identified by its device name (following your TP-... scheme).

syssi, 2025-05-06:
Let's call the project "esphome-ks-bms" and move on. I would like to avoid names of battery sellers.