syssi/esphome-tianpower-bms #31: Basen Green 48V DIY KITS V2

straga, 2024-08-07:
Anybody have information about protocol or some else.

[BASENGREEN-48V-DIY-KITS_User_Manual.pdf](https://github.com/user-attachments/files/16519075/BASENGREEN-48V-DIY-KITS_User_Manual.1.pdf)

What inside:

[image]
[image]
[image]
[image]
[image]

valdiks, 2024-08-07:
[Basengreen BMS 48200BMS-LT-55 Datasheet_V1.0doc.pdf](https://github.com/user-attachments/files/16519395/Basengreen.BMS.48200BMS-LT-55.Datasheet_V1.0doc.pdf)

straga, 2024-08-07:
> Could you start the Windows application and talk to your BMS? You can configure the protocol of the CAN bus and RS485 bus here:
> 
> ![protocols](https://private-user-images.githubusercontent.com/2735933/355704217-4d6ca6d2-ff38-4361-85de-156caeffa7f2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwMTQwNjcsIm5iZiI6MTcyMzAxMzc2NywicGF0aCI6Ii8yNzM1OTMzLzM1NTcwNDIxNy00ZDZjYTZkMi1mZjM4LTQzNjEtODVkZS0xNTZjYWVmZmE3ZjIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MDdUMDY1NjA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODlmM2YwZmRmMzBmN2FkNjE4ZWY0ZTkwNDIzZTAzNWM4MzkyODIwODE4MTdhODljOTI1NmI1OTJlNTdhZWZhMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.X25cxmujoflKQyIRVAAqhXO17a2VCZPwGsU9hu3cmRI)
> 
> Could you list all available protocols?

I Need to assembly all together and I can try sniffed protocol.

straga, 2024-08-07:
some more info, may be same https://github.com/Tertiush/bmspace

valdiks, 2024-08-07:
`[08:13:04][I][tianpower_bms_ble:389]: Mosfet status frame received
[08:13:04][D][tianpower_bms_ble:390]:   55.14.85.08.23.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AA (20)`

`[08:13:04][I][tianpower_bms_ble:340]: General info frame received
[08:13:04][D][tianpower_bms_ble:341]:   55.14.84.10.04.7A.A8.56.58.00.09.00.00.00.01.00.00.00.00.AA (20)
[08:13:04][D][sensor:094]: 'tianpower-bms-ble nominal capacity': Sending state 314.00000 Ah with 3 decimals of accuracy
[08:13:04][D][sensor:094]: 'tianpower-bms-ble capacity remaining': Sending state 221.03999 Ah with 3 decimals of accuracy
[08:13:04][D][sensor:094]: 'tianpower-bms-ble charging cycles': Sending state 9.00000  with 0 decimals of accuracy
[08:13:04][D][sensor:094]: 'tianpower-bms-ble voltage protection bitmask': Sending state 0.00000  with 0 decimals of accuracy
[08:13:04][D][text_sensor:064]: 'tianpower-bms-ble voltage protection': Sending state ''
[08:13:04][D][sensor:094]: 'tianpower-bms-ble current protection bitmask': Sending state 1.00000  with 0 decimals of accuracy
[08:13:04][D][text_sensor:064]: 'tianpower-bms-ble current protection': Sending state 'Charge overcurrent protection'
[08:13:04][D][sensor:094]: 'tianpower-bms-ble temperature protection bitmask': Sending state 0.00000  with 0 decimals of accuracy
[08:13:04][D][text_sensor:064]: 'tianpower-bms-ble temperature protection': Sending state ''
[08:13:04][D][sensor:094]: 'tianpower-bms-ble error bitmask': Sending state 0.00000  with 0 decimals of accuracy
[08:13:04][D][text_sensor:064]: 'tianpower-bms-ble errors': Sending state ''`

syssi, 2024-08-07:
@valdiks Is your battery pack / BMS already mentioned at the list of supported devices?

valdiks, 2024-08-07:
I have the same Basen DIY kit with the same BMS and equalizer models. I'm running a test now, and it looks good so far.

syssi, 2024-08-08:
@valdiks Could you tell me the advertised bluetooth name of your BMS and could you provide some more logs. In would like to lookup the `device model` (something like `TP-LTxxxx`) in your logs.

straga, 2024-08-09:
**485 Battery to Inverter**
[image]

**CAN Battery to Inverter**
[image]

valdiks, 2024-08-11:
ESPHome is showing the TP-LT55A model through MQTT.

valdiks, 2024-08-11:
We have encountered a problem where ESPHome hangs when the BLE signal is weak and only a power reset helps.

syssi, 2024-08-11:
@valdiks Could you tell me the advertised bluetooth name of your BMS?

valdiks, 2024-08-11:
@syssi TP_123456, but I've changed it.

closed 2024-08-11 by syssi