syssi/esphome-tianpower-bms #40: Issue in compiling

alex4988, 2024-11-02:
Hello,
i have tried to install your code in my esphome but i received this error:
```

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8)
 - tool-cmake @ 3.16.9
 - tool-ninja @ 1.10.2
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/config/esphome-tianpower-bms/.esphome/build/tianpower-bms-ble/.pioenvs/tianpower-bms-ble/CMakeFiles/CMakeOutput.log".
See also "/config/esphome-tianpower-bms/.esphome/build/tianpower-bms-ble/.pioenvs/tianpower-bms-ble/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /root/.platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /config/esphome-tianpower-bms/.esphome/build/tianpower-bms-ble/.pioenvs/tianpower-bms-ble/CMakeFiles/CMakeTmp

    Run Build Command(s):/root/.platformio/packages/tool-ninja/ninja cmTC_5e51a && [1/2] Building C object CMakeFiles/cmTC_5e51a.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_5e51a
    FAILED: cmTC_5e51a
    : && /root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_5e51a.dir/testCCompiler.c.obj  -o cmTC_5e51a   && :
    /root/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /root/.platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /root/.platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /root/.platformio/packages/framework-espidf/tools/cmake/project.cmake:381 (__project)
  CMakeLists.txt:3 (project)
```

Could you help me?

Best
Alessandro

syssi, 2024-11-02:
Is this a x86 platform or a different CPU architecture?

alex4988, 2024-11-02:
Arm on raspberry

syssi, 2024-11-02:
Which raspberry pi model? ESPHome requires a lot of memory at compile time. Could you try another CPU architecture? x86 should work out of the box.

alex4988, 2024-11-02:
I have a raspberry pi 4 4gb ram and os on ssd

alex4988, 2024-11-16:
I created VM on my pc and then compile from it, now i can see all parameter of my battery.

Thanks a lot for your project Syssi

closed 2024-11-16 by syssi