syssi/esphome-tianpower-bms #41: it doesn't work after a few months

italodl, 2024-11-18:
unfortunately after months of operation it stopped with the following error 

INFO ESPHome 2024.10.3
INFO Reading configuration /config/esphome/esphome-web-0e92dc.yaml...
INFO Updating https://github.com/syssi/esphome-tianpower-bms.git@main
INFO Starting log output from tianpower-bms-ble.local using esphome API
WARNING Can't connect to ESPHome API for tianpower-bms-ble.local: **Error resolving IP address: [Errno -5] No address associated with hostname (APIConnectionError)**
INFO Trying to connect to tianpower-bms-ble.local in the background

how can they solve it

syssi, 2024-11-18:
The ESP cannot be reached via network. Did you try to ping the ESO using the IP address? Is just the mDNS resolution impossible at the moment or is your ESP not available via network anymore?

italodl, 2024-11-19:
after turning everything off and on again it started working again, surely some problem with the temporary IP assignment network
thanks