syssi/esphome-tianpower-bms #43: Support for TP_BSTBD-23K-597

gxaris, 2025-02-21:
Hello all,

I was wondering if there is any chance to add support for the Basen Energy 14.3KWh 280ah battery, TP_BSTBD-23K-597.
I tried the project but unfortunately i was getting the below error:

[07:37:41][I][app:100]: ESPHome version 2025.2.0 compiled on Feb 20 2025, 21:48:23
[07:37:41][I][app:102]: Project syssi.esphome-basen-bms version 1.1.0
[07:37:41][C][wifi:600]: WiFi:
[07:37:41][C][wifi:428]:   Local MAC: D0:EF:76:48:00:20
[07:37:41][C][wifi:433]:   SSID: [redacted]
[07:37:41][C][wifi:436]:   IP Address: 192.168.150.135
[07:37:41][C][wifi:440]:   BSSID: [redacted]
[07:37:41][C][wifi:441]:   Hostname: 'basen-bms-ble'
[07:37:41][C][wifi:443]:   Signal strength: -59 dB ▂▄▆█
[07:37:41][C][wifi:447]:   Channel: 2
[07:37:41][C][wifi:448]:   Subnet: 255.255.255.0
[07:37:41][C][wifi:449]:   Gateway: 192.168.150.254
[07:37:41][C][wifi:450]:   DNS1: 192.168.150.254
[07:37:41][C][wifi:451]:   DNS2: 0.0.0.0
[07:37:41][C][logger:177]: Logger:
[07:37:41][C][logger:178]:   Max Level: DEBUG
[07:37:41][C][logger:179]:   Initial Level: DEBUG
[07:37:41][C][logger:181]:   Log Baud Rate: 115200
[07:37:41][C][logger:182]:   Hardware UART: UART0
[07:37:41][C][basen_bms_ble:523]: BasenBmsBle:
[07:37:41][C][basen_bms_ble:524]:   Fake traffic enabled: NO
[07:37:41][C][basen_bms_ble:526]: Balancing 'basen-bms-ble balancing'
[07:37:41][C][basen_bms_ble:527]: Charging 'basen-bms-ble charging'
[07:37:41][C][basen_bms_ble:528]: Discharging 'basen-bms-ble discharging'
[07:37:41][C][basen_bms_ble:530]: Total voltage 'basen-bms-ble total voltage'
[07:37:41][C][basen_bms_ble:530]:   Device Class: 'voltage'
[07:37:41][C][basen_bms_ble:530]:   State Class: 'measurement'
[07:37:41][C][basen_bms_ble:530]:   Unit of Measurement: 'V'
[07:37:41][C][basen_bms_ble:530]:   Accuracy Decimals: 3
[07:37:41][C][basen_bms_ble:531]: Current 'basen-bms-ble current'
[07:37:41][C][basen_bms_ble:531]:   Device Class: 'current'
[07:37:41][C][basen_bms_ble:531]:   State Class: 'measurement'
[07:37:41][C][basen_bms_ble:531]:   Unit of Measurement: 'A'
[07:37:41][C][basen_bms_ble:531]:   Accuracy Decimals: 3
[07:37:41][C][basen_bms_ble:531]:   Icon: 'mdi:current-dc'
[07:37:41][C][basen_bms_ble:532]: Power 'basen-bms-ble power'
[07:37:41][C][basen_bms_ble:532]:   Device Class: 'power'
[07:37:41][C][basen_bms_ble:532]:   State Class: 'measurement'
[07:37:41][C][basen_bms_ble:532]:   Unit of Measurement: 'W'
[07:37:41][C][basen_bms_ble:532]:   Accuracy Decimals: 2
[07:37:41][C][basen_bms_ble:533]: Charging power 'basen-bms-ble charging power'
[07:37:41][C][basen_bms_ble:533]:   Device Class: 'power'
[07:37:41][C][basen_bms_ble:533]:   State Class: 'measurement'
[07:37:41][C][basen_bms_ble:533]:   Unit of Measurement: 'W'
[07:37:41][C][basen_bms_ble:533]:   Accuracy Decimals: 2
[07:37:41][C][basen_bms_ble:534]: Discharging power 'basen-bms-ble discharging power'
[07:37:41][C][basen_bms_ble:534]:   Device Class: 'power'
[07:37:41][C][basen_bms_ble:534]:   State Class: 'measurement'
[07:37:41][C][basen_bms_ble:534]:   Unit of Measurement: 'W'
[07:37:41][C][basen_bms_ble:534]:   Accuracy Decimals: 2
[07:37:41][C][basen_bms_ble:535]: Capacity remaining 'basen-bms-ble capacity remaining'
[07:37:41][C][basen_bms_ble:535]:   State Class: 'measurement'
[07:37:41][C][basen_bms_ble:535]:   Unit of Measurement: 'Ah'
[07:37:41][C][basen_bms_ble:535]:   Accuracy Decimals: 3
[07:37:41][C][basen_bms_ble:535]:   Icon: 'mdi:battery-50'
[07:37:41][C][basen_bms_ble:536]: Charging states bitmask 'basen-bms-ble charging states bitmask'
[07:37:41][C][basen_bms_ble:536]:   State Class: 'measurement'
[07:37:41][C][basen_bms_ble:536]:   Unit of Measurement: ''
[07:37:41][C][basen_bms_ble:536]:   Accuracy Decimals: 0
[07:37:41][C][basen_bms_ble:536]:   Icon: 'mdi:alert-circle-outline'
[07:37:41][C][basen_bms_ble:537]: Discharging states bitmask 'basen-bms-ble discharging states bitmask'
[07:37:41][C][basen_bms_ble:537]:   State Class: 'measurement'
[07:37:41][C][basen_bms_ble:537]:   Unit of Measurement: ''
[07:37:41][C][basen_bms_ble:537]:   Accuracy Decimals: 0
[07:37:41][C][basen_bms_ble:537]:   Icon: 'mdi:alert-circle-outline'
[... 235 lines cut ...]
[07:37:43][C][ble_client:027]: BLE Client:
[07:37:43][C][esp32_ble_client:048]:   Address: E8:8D:A6:36:50:A0
[07:37:43][C][esp32_ble_client:049]:   Auto-Connect: TRUE
[07:37:43][C][esp32_ble_client:083]:   State: ESTABLISHED
[07:37:43][C][mdns:116]: mDNS:
[07:37:43][C][mdns:117]:   Hostname: basen-bms-ble
[07:37:43][C][esphome.ota:073]: Over-The-Air updates:
[07:37:43][C][esphome.ota:074]:   Address: basen-bms-ble.local:3232
[07:37:43][C][esphome.ota:075]:   Version: 2
[07:37:43][C][safe_mode:018]: Safe Mode:
[07:37:43][C][safe_mode:020]:   Boot considered successful after 60 seconds
[07:37:43][C][safe_mode:021]:   Invoke after 10 boot attempts
[07:37:43][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[07:37:43][C][api:140]: API Server:
[07:37:43][C][api:141]:   Address: basen-bms-ble.local:6053
[07:37:43][C][api:143]:   Using noise encryption: YES
[07:37:44][W][basen_bms_ble:226]: [E8:8D:A6:36:50:A0] Not connected
[07:37:54][W][basen_bms_ble:226]: [E8:8D:A6:36:50:A0] Not connected
[07:38:04][W][basen_bms_ble:226]: [E8:8D:A6:36:50:A0] Not connected

Thank you.

[image]

[image]

gxaris, 2025-02-21:
Some more logs that show the issue:

[07:45:30][C][esp32_ble_tracker:686]:   Connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[07:45:30][D][esp32_ble_client:154]: [0] [E8:8D:A6:36:50:A0] Found device
[07:45:30][D][esp32_ble_tracker:700]: Found device E8:8D:A6:36:50:A0 RSSI=-30
[07:45:30][D][esp32_ble_tracker:721]:   Address Type: PUBLIC
[07:45:30][D][esp32_ble_tracker:723]:   Name: 'TP_BSTBD-23K-597'
[07:45:30][D][esp32_ble_tracker:231]: Pausing scan to make connection...
[07:45:30][C][ble_client:027]: BLE Client:
[07:45:30][C][esp32_ble_client:048]:   Address: E8:8D:A6:36:50:A0
[07:45:30][C][esp32_ble_client:049]:   Auto-Connect: TRUE
[07:45:30][C][esp32_ble_client:083]:   State: DISCOVERED
[07:45:30][D][esp32_ble_tracker:116]: connecting: 0, discovered: 1, searching: 0, disconnecting: 0
[07:45:30][I][esp32_ble_client:111]: [0] [E8:8D:A6:36:50:A0] 0x00 Attempting BLE connection
[07:45:30][C][mdns:116]: mDNS:
[07:45:30][C][mdns:117]:   Hostname: basen-bms-ble
[07:45:30][D][esp32_ble_tracker:116]: connecting: 1, discovered: 0, searching: 0, disconnecting: 0
[07:45:30][C][esphome.ota:073]: Over-The-Air updates:
[07:45:30][C][esphome.ota:074]:   Address: basen-bms-ble.local:3232
[07:45:30][C][esphome.ota:075]:   Version: 2
[07:45:30][C][safe_mode:018]: Safe Mode:
[07:45:30][C][safe_mode:020]:   Boot considered successful after 60 seconds
[07:45:30][C][safe_mode:021]:   Invoke after 10 boot attempts
[07:45:30][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[07:45:30][C][api:140]: API Server:
[07:45:30][C][api:141]:   Address: basen-bms-ble.local:6053
[07:45:30][C][api:143]:   Using noise encryption: YES
[07:45:31][D][esp32_ble_client:154]: [0] [E8:8D:A6:36:50:A0] ESP_GATTC_CONNECT_EVT
[07:45:31][D][esp32_ble_client:154]: [0] [E8:8D:A6:36:50:A0] ESP_GATTC_OPEN_EVT
[07:45:31][D][esp32_ble_tracker:116]: connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[07:45:31][D][esp32_ble_tracker:282]: Starting scan...
[07:45:32][D][esp32_ble_client:352]: [0] [E8:8D:A6:36:50:A0] Event 46
[07:45:32][D][esp32_ble_client:154]: [0] [E8:8D:A6:36:50:A0] ESP_GATTC_SEARCH_CMPL_EVT
[07:45:32][I][esp32_ble_client:273]: [0] [E8:8D:A6:36:50:A0] Connected
[07:45:32][E][basen_bms_ble:135]: [E8:8D:A6:36:50:A0] No notify service found at device, not an BASEN BMS..?
[07:45:32][D][switch:055]: 'basen-bms-ble enable bluetooth connection': Sending state ON
[07:45:32][D][esp32_ble_client:234]: [0] [E8:8D:A6:36:50:A0] cfg_mtu status 0, mtu 517
[07:45:38][W][basen_bms_ble:226]: [E8:8D:A6:36:50:A0] Not connected
[07:45:48][W][basen_bms_ble:226]: [E8:8D:A6:36:50:A0] Not connected
[07:45:58][W][basen_bms_ble:226]: [E8:8D:A6:36:50:A0] Not connected

syssi, 2025-02-21:
Could you share the link to the android app you are using?

gxaris, 2025-02-21:
> Could you share the link to the android app you are using?

Sure, I have uploaded it here:
https://drive.google.com/file/d/1OHLEG1ebBA6yGOwYcIlhO1p1HuYb8Cod/view?usp=drive_link

syssi, 2025-02-21:
Keep in mind: The BLE module of the BMS is able to handle a single connection only.

closed 2025-02-21 by syssi