syssi/esphome-tianpower-bms #46: Tianpower TP_BSTBD-23J-512 not supported

ThetaGamma, 2025-04-08:
Hi there,
I tried the tool with my 300Ah 14.3kW Basen manufactured 2023. It announces itself as via BT as 

```
[19:14:53][I][ble_adv:040]: New Tianpower found
[19:14:53][I][ble_adv:041]:   Name: TP_BSTBD-23J-512
[19:14:53][I][ble_adv:042]:   MAC address: DC:XXXXXXXXX
```
(output from esp32-ble-scanner, pls notice the "J" in the string)

I recently updated Firmware with HY-BT200LT55-V1.0.4-T3.
BT connection with iPhone works with Basenenergy App, but is pretty unstable.

connection with esp32-ble-example is not working with error
`[19:25:13][W][tianpower_bms_ble:217]: [90:A6:BF:93:A0:69] Not connected
`
any idea how to make the tool working?
Cheers TG

syssi, 2025-04-08:
At the BLE scanner log above the BMS is advertised with a `DC:...` MAC address. The second log show a BLE client connection to a different MAC address. Did you change the YAML configuration to the MAC address of your BMS?

ThetaGamma, 2025-04-08:
sorry, no. Now I did and it works perfectly. Thanks!
I didn't use the "external component" way, but the CLI. May I suggest to add this step of changing the MAC address in the README Installation section for other noobs like me? ;-)

closed 2025-04-08 by ThetaGamma