syssi/esphome-tianpower-bms #53: not sure what I'm doing wrong installing this

dlarue, 2025-06-15:
I installed HA via virtualBox vm, forked the repo and added secrets.yaml including my 3 BasenGreen BMS Mac addr:

bms0_mac_address: 00:D6:CB:80:81:74
bms1_mac_address: 00:D6:DB:80:82:BB
bms2_mac_address: 98:A1:4A:52:C3:EF

wifi_ssid: !secret wifi_ssid
wifi_password: !sectet wifi_password

mqtt_host: !secret mqtt_host
mqtt_username: !secret mqtt_username
mqtt_password: !secret mqtt_password

Then after installing the ESPHome Builder in HA I created a test device then added the external component entry:

external_components:
- source:
      type: git
      url: http://github.com/dlarue/esphome-tianpower-bms
      ref: main

The creation of the test device created API key, OTA and WiFi entries and once built, downloaded then installed I can see in the log window the device starts, connects to wifi and has a network connection but I don't know how to see any data. Sorry, HA noob.
Any links or ideas what I'm missing?

dlarue, 2025-06-15:
I'm not sure what "start with a single device/BMS means but I'm guessing it just starting with a device I have communications with( ESPHome- Test ) and then do an OTA update using one of the YAML files like esp32-ble-example.yaml with one of my Mac addresses entered. I will see if that works or if I have to replace the default YAML file of the device with the example one.
Thanks, giving it a go now.

syssi, 2025-06-16:
You have two options to push the measurements to Home Assistant: The native `api` or `mqtt. Did you enable the ESPHome integration in Home Assistant and did you add the ESPHome node? This is your first ESPHome node, right?

dlarue, 2025-06-16:
Sounds like what I've done, ie left MQTT data commented out of the YAML file, I found the tianpower-bms-ble.local host using the ESPHome integration after building and installing the device using the ESPHome Device Builder in Home Assistant.
I see all 3 of my battery packs listed in one long 'card' on the Home Assistant dashboard. It's clipping off descriptions and it looks like the values are all integers instead of floating point values. Therefore, I find the web page emitted by the device easier to look at and read.

syssi, 2025-06-16:
Could you go to:

```
Settings -> Devices -> ESPHome -> "device name" + 1 device and x entities
```

and provide a screenshot of all entities of the selected ESPHome node (device)?

dlarue, 2025-06-16:
do you mean like this?  If so, the entities list is 154 items long containing 3 BMSs.

[image]

syssi, 2025-06-16:
Yes. I'm talking about this view. It looks like all measurements are properly available. Just the resolution of some measurements doesn't look good. Could you try to edit the `Device0 cell voltage 1` sensor entity and change the resolution?

dlarue, 2025-06-16:
I found something, it looks like it's a problem with the latest version of HA where 2025.5.2 works fine but 2025.6.1 has some values showing no decimal point precision. 
https://github.com/home-assistant/core/issues/146873

Sleeper85, 2025-06-16:
The default display precision is not respected.

[image]

[image]

closed 2025-06-16 by dlarue