syssi/esphome-tianpower-bms #54: HA default precision changed to zero so most values are shown as integers

dlarue, 2025-06-17:
As of HA v2025.6.0 there was a change to the default displayed precision which used to be 3 digits and now the default is 0 digits.
What's worst is that the system used to accept "accuracy_decimals:" but now that is disregarded and only  suggested_display_precision works.  "suggested_display_precision: 3"

It's a royal PIA to go through the display for each entity and change the display precision manually.

syssi, 2025-06-17:
I guess this will be improved pretty soon.

dlarue, 2025-06-17:
It does not look like HA will be changing anything to "fix" this. They made the conscious decision to change the default, to disregard the existing setting of "accuracy_decimals" and to add "suggested_display_precision".  So everyone needs to add suggested_display_precision to their YAML files or else many numbers will look like integers. :(

syssi, 2025-06-17:
Let's keep this issue open for a another day. I will like to verify the situation can or cannot be improved.

syssi, 2025-06-22:
This issue must be fixed at the ESPHome integration of HA core in my opinion. See https://github.com/home-assistant/core/issues/146873

dlarue, 2025-06-22:
It's really a PIA having to manually change all the display elements to fix this poor behavior. Hoping version 1.7 fixes the regression but feel there should be a point release update.

Macrisu, 2025-06-29:
I've just upgraded the esphome integration and nothing.
I have 3 ways to read my BMS values, all with the same display problem!

Frustrating.

syssi, 2025-07-03:
This is the pull request to resolve the issue: https://github.com/home-assistant/core/pull/147849/files

It's part of HA core 2025.7.

closed 2025-07-03 by syssi