syssi/esphome-tianpower-bms #58: Would it be possible to add if balancing is ongoing ?

jjtoard, 2025-06-29:
Would it be possible to add if balancing is ongoing ?

syssi, 2025-06-29:
Could you share some screenshots of the PC software? Are you able to provide some RAW traffic from the PC software?

syssi, 2025-06-29:
I've tried to add the feature. Please give it a try by updating your configuration YAML to the feature branch:

```
substitutions:
  name: tianpower-bms-ble
  device_description: "Monitor a Tianpower Battery Management System via BLE"
  external_components_source: github://syssi/esphome-tianpower-bms@add-balancer
  mac_address: !secret bms0_mac_address
```

And add the new sensors to your YAML. See https://github.com/syssi/esphome-tianpower-bms/pull/59/files#diff-6d19e7b8b7435fb5c57e1ad1b8498afa48c2073ed985bf1fc1d82f08eb2ded56

closed 2025-06-29 by syssi